LibrePCB/LibrePCB #1364: STEP exporting is missing a "BOARD".

xotho, 2024-04-04:
```
LibrePCB Version: 1.1.0
Git Revision:     18a3d4589
Build Date:       2024-04-03T09:50:18
Build Author:     LibrePCB CI
Qt Version:       6.6.2 (built against 6.6.2)
CPU Architecture: x86_64
Operating System: Windows 11 Version 23H2
Platform Plugin:  windows
TLS Library:      OpenSSL 3.0.12 24 Oct 2023
OCC Library:      OCCT 7.7.2
...
Exported the STEP of a double board, and importing in FREECAD or PTC one board is missing.
...
The two boards in the 3D view of LIBREPCB
![image](https://github.com/LibrePCB/LibrePCB/assets/2029366/9438d450-22dc-4646-8926-8b3f372e0975)

The imported Boards from FREECAD.
![image](https://github.com/LibrePCB/LibrePCB/assets/2029366/02f84552-65a3-4e53-9943-99f0a9f44d74)

take care,
X!

ubruhin, 2024-04-04:
Note that generally it's not recommended and very unusual to have multiple board outlines as we don't know if and how PCB manufacturers can handle that. It might lead to problems or questions from the manufacturer, thus also a DRC warning is emitted for such a project. It's recommended to create a separate LibrePCB project for each PCB.

However, I agree that the STEP export should behave the same way as the integrated 3D viewer :+1:

xotho, 2024-04-04:
With a product composed by two PCBs soldered at 90 degrees with a dimension of  A (10.71mm x 14.92mm) + B (10.71mm x 2.54mm)  ... on the same BOARD,  I think it's a must at least for me even if uncommon (agree).

Thank you.
X!

ubruhin, 2024-04-04:
Well, it doesn't matter what is the use-case or whether it is more intuitive this way or whether we would like to support this workflow or not - what PCB manufacturers will do with such a design is not under our control. You can do it, but you need to accept the risk of wrong production. Thus the DRC warning.

Of course things are a bit different when you produce PCBs on your own, then you can do whatever you want as you have production under control.

xotho, 2024-05-21:
thank you, tested and working.

take care,

PR #1392 (2024-05-24): STEP export: Include all PCB outlines, if multiple
For consistent behavior with the integrated 3D viewer, export all board outlines to the STEP file.

Fixes #1364
closed 2024-05-24 by ubruhin