LibrePCB/LibrePCB #1371: "Settings..." Menu-Item in App-Menu Changes to "Preferences..."

sparkylein, 2024-04-07:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 1.1.0
Git Revision: 
Qt Version:
Operating System: macOS 14.4.1
...
```

##### SUMMARY

"Settings..." Menu-Item in App-Menu Changes to "Preferences..."

##### STEPS TO REPRODUCE

1) Start LibrePCB 
2) Click on "Settings..." menu item of App-Menu, Workspace Settings Dialog opens
3) Click either on "Cancel" or "OK" button in Workspace Settings Dialog
4) The "Settings..." Menu-Item in App-Menu will change to "Preferences..."

##### EXPECTED RESULTS

Consistency in menu item name (if not changed on purpose, which in this case I don't think it is...)

##### ACTUAL RESULTS

"Settings..." Menu-Item in App-Menu changes to "Preferences..." after closing the Workspace Settings dialog the 1st time.

##### ADDITIONAL INFORMATION

On macOS this menu item should be "Preferences..." and maybe on other OS "Settings...".
Either way this menu item name should be consistent not switch around.

ubruhin, 2024-04-07:
Hm this sounds like a Qt bug, I don't think the rename is somewhere in our code :thinking: 

Could you please fill out the whole "VERSION / OS / ENVIRONMENT" section above? In the "About LibrePCB" dialog this information can be copied with one click. It's important to know because the Qt version is stated there.

sparkylein, 2024-04-07:
Done.

ubruhin, 2025-08-04:
The new UI (see #1494) does not use Qt windows anymore, so I suspect this issue can be closed. Unfortunately at the moment the native macOS menubar is not used anymore by the new UI, but I think sooner or later we will re-introduce it.