LibrePCB/LibrePCB #1372: App-Menu Items "About LibrePCB" and "About Qt" of the Control Panel disappear

sparkylein, 2024-04-07:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 1.1.0
Git Revision:
Qt Version:
Operating System: macOS 14.4.1
...
```

##### SUMMARY

Under certain circumstances the App-Menu Items "About LibrePCB" and "About Qt" (while the Control Panel has the focus) disappear, but come back again when a editor window is opened again.

##### STEPS TO REPRODUCE

1. Start LibrePCB and the Control Panel Window appears (until now everything is okay)
2. Open a project so a Schematic Editor Window and a Board Editor Window appear
3. Close both Editor Windows with the macOS "Close Window Button" (Round & Red) not the Big [Square & Red] Qt button.
4. App-Menu Items "About LibrePCB" and "About Qt" of the Control Panel Window disappeared

##### EXPECTED RESULTS

The "About LibrePCB" and "About Qt" Menu-Items of the Control Panel Window should not disappear 

##### ACTUAL RESULTS

App-Menu Items "About LibrePCB" and "About Qt" of the Control Panel Window disappeare and the "Preferences..." Menu-Item is the top menu item.

##### ADDITIONAL INFORMATION

[image]

ubruhin, 2024-04-07:
Same here, could you fill out the whole "VERSION / OS / ENVIRONMENT" section above? Sounds like a Qt bug too.

sparkylein, 2024-04-07:
Done.