LibrePCB/LibrePCB #1374: Library Editor: {{NAME}} not considered correctly in the schematic

Epyco, 2024-04-12:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 1.1.0
Git Revision:     18a3d4589
Build Date:       2024-04-03T09:50:18
Build Author:     LibrePCB CI
Qt Version:       6.6.2 (built against 6.6.2)
CPU Architecture: x86_64
Operating System: Windows 10 Version 22H2
Platform Plugin:  windows
TLS Library:      OpenSSL 3.0.12 24 Oct 2023
OCC Library:      OCCT 7.7.2
```

##### SUMMARY
If I create a component and add an attribute NAME to it, the name is not displayed correctly in the schematic.

##### STEPS TO REPRODUCE
- Create a symbol "Issue" with just the {{NAME}} and {{VALUE}}
- Create a component "Issue", link the symbol "Issue" to it and keep the rest as default or empty
	- Add an attribute:
		Key: NAME
		Type: String
		Value: SomeName
	
	- Add an attribute:
		Key: VALUE
		Type: String
		Value: SomeValue
- Create a footrpint "Issue" with just the {{NAME}} and {{VALUE}}
- Create a device "Issue" and link the footprint and component to "Issue"

Place the component in the schematic.

##### EXPECTED RESULTS
> SomeName

##### ACTUAL RESULTS
> ?1

##### ADDITIONAL INFORMATION

ubruhin, 2024-04-24:
Hmm, why do you want to add a `NAME` attribute? This is a built-in attribute and makes no sense to overwrite manually as it is very critical (needs to be unique etc.)

Epyco, 2024-04-25:
Sorry, but unfortunately I no longer know what I wanted to achieve or implement with it.

closed 2024-04-25 by Epyco