LibrePCB/LibrePCB #1387: STEP export issue

xotho, 2024-05-08:
##### VERSION / OS / ENVIRONMENT
LibrePCB Version: 1.1.1-unstable
Git Revision:     4d05a631f
Build Date:       2024-05-06T08:58:42
Build Author:     LibrePCB CI
Qt Version:       6.6.2 (built against 6.6.2)
CPU Architecture: x86_64
Operating System: Windows 11 Version 23H2
Platform Plugin:  windows
TLS Library:      OpenSSL 3.0.12 24 Oct 2023
OCC Library:      OCCT 7.7.2

##### SUMMARY
Exported 3D-STEP show a PCB where a component mounted on BOTTOM side of the PCB turned 180 degree.
In the LibrePCB all is shown correctly instead.

Wrong location about exported: 
[image]

Right position shown in LibrePCB:
[image]

ubruhin, 2024-05-10:
I don't understand from the pictures what is wrong. It doesn't even look like the same PCB is shown in both pictures :thinking: 

Can you please post the whole project (or send it to the contact email address on the website)?

xotho, 2024-05-10:
Dear ubruhin,

check the following images, ( I removed some components ):

1)
[image]

exported and imported in FREECAD or PTC:

2)
[image]

1) is the right view in LIBREPCB, 2) is the wrong view imported in FREECAD or PTC.

Should be clear I think.

Do you need more ?

ubruhin, 2024-05-14:
I still cannot reproduce this, it would be very helpful to have your project for investigation. If you send it (as *.lppz export) to the contact email on our website, I'll handle it confidentially.

xotho, 2024-05-14:
Pls Check **contact** Email.

take care

ubruhin, 2024-05-14:
Got it, and can reproduce it! Thank you!

xotho, 2024-05-14:
Great. Thank you 2 U.

ubruhin, 2024-06-11:
OK this is really strange... The affected STEP model is a bit unusual because it contains a rotation of 180°, as shown in the FreeCAD object inspector:

[image]

But of course also this unusual case needs to be supported properly by LibrePCB. However, after exporting the PCB to STEP, this rotation is lost on the package:

[image]

I don't know why this happens, but I realized it can be fixed with a small change (see #1401). The output structure is then as expected:

[image]

And even the label "DEV1:1" is now correct, although this makes absolutely no sense to me. Also the change in #1401 makes no sense to me, but at least I didn't experience any negative impact on several projects I tested. That's the mystery of the OpenCascade API which I still don't understand at all...

I think we can go with this change for now and if any new issue pops up in future we have to investigate it again.

@xotho would be nice if you could test your project with the nightly build of #1401 (should appear in ca. an hour) and report here if the exported STEP is now correct.

PR #1401 (2024-06-14): STEP export: Fix possibly wrong placement of devices
See comment in source code and discussion in #1387.

Fixes #1387.
closed 2024-06-14 by ubruhin