LibrePCB/LibrePCB #1388: PCB rotate

xotho, 2024-05-14:
The ROTATION do not ROTATE a complete PCB with component in the right way.

##### STEPS TO REPRODUCE
TURN ON ALL LAYER.
DRAW a RECTANGULAR SELECTION AREA to select all the elements interested in the function.
EDIT -> ROTATE

##### EXPECTED RESULTS
A ROTATION of 90degrees of ALL elements.

##### ACTUAL RESULTS
ROTATION follow strange rules, effect is LAYER/component are not aligned in the rotation. TOP and BOTTOM COPPER is ROTATED , but some components are not moved.... that's actually a mess.

take care,

ubruhin, 2024-05-14:
Did you lock some component's placement? You might try temporarily enabling "View -> Ignore Placement Locks".

xotho, 2024-05-14:
Yes , I had some components in locked status.

The suggested: "**View -> Ignore Placement Locks**" , solved the issue.

Just one logic consideration:
if i have locked components, but i rotate a full workspace or a complete PCB (maybe one of the PCBs I can have in the workspace),  should not have I to expect a complete PCB rotate in sync with the locked components?

or request the user if he wants to " Ignore Placement Locks"  after having launched the command ?

take care, thank you again

ubruhin, 2024-05-14:
Yeah I'm not sure yet how to handle this situation - sometimes this is annoying, but it's the purpose of the locks to behave like that. Maybe a hint "some locked components are selected" with the option to ignore their locks would make sense.

xotho, 2024-05-14:
Agree

AvetosDesign, 2026-09-18:
Would overlaying a "prohibited" symbol (red circle with a slash) on each locked device/entity be an acceptable approach?

ubruhin, 2026-09-18:
Somehow I'm thinking of a message "tooltip" which appears when trying to drag (partially) locked objects:

[image]

The message may disappear automatically after a timeout. And it may contain a button to temporarily ignore locks (same button as in the status bar). It should also appear e.g. when the user clicks "r" to rotate the selected objects, if some of them are locked.

The idea behind this would be to keep the UI clean as long as the user doesn't try to drag locked objects (in contrast e.g. to a permanent "prohibited" symbol on each device, which clutter the UI). But we may have to try it out to see its look&feel.

AvetosDesign, 2026-09-19:
We can go back to a simple text message, like you show above, if you prefer it.  Would you mind trying the branch I just committed on my fork first, though?  It displays a small lock icon over locked items, but only when you try to move (translate, rotate, flip...) them.  The icons disappear after a short period of time (750 mS).  I don't think it's too obtrusive, but I would appreciate your feedback on the approach.  The change touches only four files -- the editor's graphicsscene.cpp/.h and the fsm boardeditorstate_select.cpp/.h.

AvetosDesign, 2026-10-12:
Here's a short video of the concept:
https://github.com/user-attachments/assets/9b9dcd9b-40c9-4e7b-8618-0b532d563ff3