LibrePCB/LibrePCB #1390: Add auto dark mode for Windows

mi4code, 2024-05-19:
##### SUMMARY

[image]

##### SOLUTION

I can open PR for my current implementation if desired but I think that additional settings entry should be added to have option to disable this feature (and I have no idea how to do that so I will need some help).

ubruhin, 2024-05-20:
Hi,

can you quickly describe how you implemented that? Or feel free to open a PR with your current implementation. I'm not sure yet if and how this feature should be configurable, first I'd like to test the look&feel on my own. Also I'm curious what you mean with "auto", does Qt detect if Windows is set to a dark mode or is there any manual detection required?

ubruhin, 2024-05-21:
OK thanks for the PR. So basically everything is manually done (detecting Windows configuration & applying different colors).

I found [here](https://www.qt.io/blog/dark-mode-on-windows-11-with-qt-6.5) that Qt now has a dark theme built-in, which can be tried by passing `-style fusion` to the `librepcb.exe`. It only works with our 64-bit (Qt6) build, but that's fine as 32-bit (Qt5) builds are deprecated.

I quickly tried this and it seems to work very well. So probably the only thing is to detect the Windows theme setting, and if dark then enable the Fusion style. No need for changing colors manually.

Still not sure about configurability... maybe for now it's good enough to ignore it. If the user selected dark theme, he is probably not interested in light theme of LibrePCB :thinking:

mi4code, 2024-05-21:
> I quickly tried this and it seems to work very well. So probably the only thing is to detect the Windows theme setting, and if dark then enable the Fusion style. No need for changing colors manually.

Actually the Fusion style setting already does this (it uses dark/light theme according to the system) so using it on Windows by default would solve everything.

Only downside would be that the light theme will become slightly different - see comparsion below (left current, right fusion).

[image]

ubruhin, 2024-05-30:
> Only downside would be that the light theme will become slightly different - see comparsion below (left current, right fusion).

Yeah although the difference is rather small, IMHO the default light theme still looks a bit better. Thus I added a conditional to #1391 to use Fusion style only if Windows is set to dark theme. In a quick test this seems to be working well. And because I don't like to make this configurable in workspace settings (yet), I added the possibility to disable dark theme with an environment variable - just in case it doesn't work properly for some users, generally I intend to remove this fallback as soon as we are confident that the dark mode works well.

closed 2024-06-01 by ubruhin