LibrePCB/LibrePCB #1393: Automatically re-number all components in project

ubruhin, 2024-05-21:
##### SUMMARY

Many iterations of modifications in a schematic can make component names (R1, R2, R3, ...) to become unordered and thus unintuitive and feeling "messy". Thus it would be handy to allow re-numbering all components automatically.

Initial feature request: https://librepcb.discourse.group/t/things-i-found-difficult-or-unintuitive-as-a-newcomer/747

##### SOLUTION

In the schematic editor, probably just a new menu item is needed in the main menu, with a confirmation dialog which quickly explains what will happen. Then assigning all components of the project depending on their location - starting from the first page to the last page, from top to bottom, left to right. Maybe a more sophisticated solution could be chosen, e.g. respecting groups of components but that's probably too much magic :thinking:

RDamman, 2024-07-13:
I think renumbering should be implemented as a non modal dialog windows. Here you have:
- input field **index number**: the next number to be assigned after a component prefix.
- selection list **conflicting component names:**  Options: Skip / Reassign / Ask (Skip/Reassign).
- checkbox (don't) **increment index number when same component clicked again**.
- Close button.

Behaviour:
On renumbering the dialog shows. The index number is by default 1 or the index number of the component selected. The user clicks the component. The component gets the current dialog index number assigned and the dialog index number is incremented by 1. The user clicks the next component. When a user wants to skip 10 numbers the user can enter the desired number in the input field with the index number.

moshee, 2024-07-26:
Tagging onto this to add that when copy/pasting multiple components, the pasted components' refdes are randomly assorted, which is not tidy. I would propose applying the same refdes assignment logic to be developed here to pasted components.

I have shared this same screenshot in other threads because this one task made me notice multiple areas of improvement:

[image]

First row was placed manually, others were copy pasted from first row.

ubruhin, 2024-07-26:
> Tagging onto this to add that when copy/pasting multiple components, the pasted components' refdes are randomly assorted, which is not tidy. I would propose applying the same refdes assignment logic to be developed here to pasted components.

Oh that's a good point which I didn't realize yet - could you maybe open a separate issue for that as it is not directly related to this one? Basically it's a "bug" of the paste operation, not a new feature...

closed 2026-11-03 by ubruhin