LibrePCB/LibrePCB #1405: STEP EXPORTING EXPORT PARTIALLY COLORS

xotho, 2024-06-14:
I could not check in deep, but I think there is something lost (colors) when exporting to STEP.

In BOARD EDITOR ( 3D view ) you can see the imported colors from the models but once exported some colors are lost.

Later I will provide images..

take care,
X!

ubruhin, 2024-06-15:
I saw this problem several times but it was never the issue of LibrePCB. Sometimes using a different FreeCAD version helps (so the STEP is actually fine) and sometimes FreeCAD shows the colors for 0.1sec after opening the model and then they everything changes to gray. So again it's FreeCAD which refuses to show correct colors.

I'll close this issue for now therefore. If you have any evidence that it's the fault of LibrePCB, feel free to reopen.

P.S. I don't know why you use uppercase letters so often (even the whole issue title), but it is a bit disruptive. Would be nice to write normal sentences with normal capitalization :wink: Thanks.

closed 2024-06-15 by ubruhin

xotho, 2024-06-15:
That is not a real problem for sure,  anyhow I saw the same question using PTC.

I will recheck next week so just to give you a feedback.

I think I lost something, what's the standard version that LibePCB manage in import as STEP file ?

take care.

ubruhin, 2024-06-15:
> I will recheck next week so just to give you a feedback.

Perfect :+1:

> I think I lost something, what's the standard version that LibePCB manage in import as STEP file ?

Hmm not sure if I understand correctly, you mean what STEP versions? We use OpenCascade for reading STEP files and according [this page](https://old.opencascade.com/content/data-exchange) it supports AP203, AP214, AP242.

xotho, 2024-06-15:
Yes, that's it.

Do u wanna me to check the named 3 standard formats possibly ?

thank you.

ubruhin, 2024-06-15:
AP203 does not support colors so you don't need to check it. AP242 and AP214 might make sense to test both :+1:

xotho, 2024-06-15:
ok