LibrePCB/LibrePCB #1414: Add option to show wire overlaps/crossovers as jumps

flotwig, 2024-07-13:
##### SUMMARY

Currently, wires that overlap/cross over each other without connecting are displayed as line segments. Despite the fact that connections between wires are shown with a circle, it can be difficult to tell at a glance if two wires are connected or not. Example:

[image]

##### SOLUTION

It would be nice to have the option to display non-connecting wires that cross over each other as "jumps", with a curved line. This would help to tell at a glance if two wires should connect. Example:

[image]

ubruhin, 2024-07-13:
Really not sure how this could be implemented (not as trivial as it sounds) but I totally agree this would be nice :+1:

xotho, 2024-07-16:
Why not use just DOT on every wire connected together and no DOT for all the overlaps/crossovers?

So all connection including wire to IC's and so on should always have a DOT if connected together....

Exemples:
[image]

[image]

[image]

take care,
X!

flotwig, 2024-07-16:
> Why not use just DOT on every wire connected together and no DOT for all the overlaps/crossovers?
> 
> So all connection including wire to IC's and so on should always have a DOT if connected together....

@xotho This seems to be how LibrePCB currently displays connected/unconnected crossovers, but it can be difficult to read at a glance (in my opinion). Please let me know if I misinterpreted what you were saying.

ubruhin, 2024-07-17:
> Why not use just DOT on every wire connected together and no DOT for all the overlaps/crossovers?
> So all connection including wire to IC's and so on should always have a DOT if connected together....

IMHO that would be very distracting and also does not conform to how schematics (have to) look.

It's already clear anyway which lines are connected or not because LibrePCB automatically adds the dot on junctions with more than 2 legs. The suggestion of @flotwig would just make it even more clear.

xotho, 2024-07-17:
Yes Flotwig, as it is now it is ok from my private point of view. The only thing that I would consider is to show DOT at device connection level anyway.

**Actual implementation** where A are not shown as DOT....
[image]

Device connection level with DOT
[image]

An then, in the **Actual implementation** I would also change color of the annular connection (B) to show lines that have to be connected or with optional connection .... or other classes still to be defined .....

Personally the "Jumps" are not a good solution due to the effort in development and the dimension .... when zoomming-in and zoomming-out and it should be an element that take off attention when checking (for me), but I understand the request.

Take care,
X!

giancarlo64, 2024-08-04:
I personally think that the "semicircle" to indicate that there is no connection is only an graphical and aesthetically solution and has nothing to do with a electrical circuit. If you have more than 1 line to cross it looks horrible.
[image]

I think that electrical diagrams made in this way are unwatchable. 

but that's just my opinion

Gian Carlo

xotho, 2024-08-05:
Why not adding a set-up option to toggle between the two ?

Take care
X!