LibrePCB/LibrePCB #1416: Pins/wires do not connect after moving

moshee, 2024-07-26:
##### SUMMARY

Apologies if this has been reported already. It seems very basic but I could not find it.

Wires/pins do not automatically connect together when their loose ends are moved together. For example in the following edits, or when pasting wire segments.

[image]

This seems like it may have been an intentional design decision to avoid accidental connections. However, it has some bad effects:
* Fake junctions that do not actually connect unless you notice there is no junction dot
* Wires in repetitive circuits must be completely drawn one by one
* Wires must often be deleted and re-drawn to add new connections to a design to avoid adding a bunch of invisible nodes to the wire.

##### SOLUTION

Every other ECAD tool I have used will either

1. Connect loose ends together (Xpedition, Altium), or
2. "normalize" wires so that drawing a short segment to connect a loose end to something will not leave an invisible node in the middle of the wire (DipTrace).

The shortcoming described in (2) will hopefully be alleviated by #710 if it is also applied to the schematic editor. However, I find (1) to be greatly desirable. A very obvious indication that two nets/entities are about to be merged before mouseup/confirm would be appropriate.

Some logic would need to be implemented to decide who wins if two different nets are about to be connected. As a starting point, maybe a heuristic that combines named net status (power port or net label on a net gives it higher weight) with total number of nodes in the net could be applied. Ideally this behavior would be as stable and predictable as possible (not random). The net can always be renamed later if the tool went against the user's wishes.

ubruhin, 2024-07-26:
Yes basically it is intended that move operations do not connect wires because in many cases this is not desired and can be very annoying. But I agree sometimes it's intended to connect wires and thus the editor should allow it, so I plan to implement that. It only needs to be decided *how* exactly the logic should be (e.g. enable/disable autoconnect with a keyboard modifier, or ask with a context menu whether to connect moved elements or not).

And yes, #710 will improve the situation too.

RDamman, 2024-07-26:
I think in the schematic editor it is almost always desirable that wires become connected. In case of a accidentally connection you can simply undo the action. In what cases should it not be desirable?

ubruhin, 2024-07-26:
> In which cases should it not be desirable?

I just remember that Altium annoyed me sometimes with creating connections I didn't want. Especially when just moving something around temporarily. Maybe it's not the best example, but consider this one:

[image]

When moving J2 1 grid to the left, it will connect pin 7 to GND, possibly even without you notice it. And moving it back to the right won't disconnect it. Somehow for me it feels unintuitive (and a bit dangerous) if the tool creates connections I didn't ask for.

Anyway, I think in the end we need to make some experiments what works best. I can't say yet whether auto-connect should be opt-in or opt-out.

RDamman, 2024-07-26:
Yes, you're right when moving a component. I think nobody wants that. But what when you consider the case of (unconnected) wire endpoints solely? I think in that case you would make nearly always want to make a connection.

ubruhin, 2024-07-26:
Let's clarify all the details when this feature is getting implemented :wink: I'm sure we'll find a good solution then.

moshee, 2024-07-26:
I think it is reasonable that dragging a pin onto a wire segment (not loose end) would not automatically create a connection. This is purely opinion but I don't like when a wire connected to a pin immediately turns 90 degrees without a straight segment first anyways.

However, in most other edit cases (loose end to segment, loose end to loose end, loose end to pin), it is more of a hindrance than a help to suppress connection.

In any case, having a clear counter-indication (instead of the lack of a junction dot) would be helpful to indicate any situations that look like junctions but are not. There are existing glyphs for wire ends and pins, but they are faint especially when things overlap in a false junction.

MvDDD, 2025-02-04:
maybe a shortcut that resolves touching connectors and optimizes wires like this [image]
**within the current selection** could solve most of these problems