LibrePCB/LibrePCB #1420: Incomplete grid area shown if applied a scale factor to graphic drivers.

xotho, 2024-07-29:
Windows 11 Pro platform.
Quadro RTX4000 ADA, drivers v555.99

As per title , here follow image examples : 

[image]

[image]

Applied scaling factor : 
 
[image]

take care ,
X!

xotho, 2024-07-29:
Issue not present in this version:

[image]

Sorry,
X!

ubruhin, 2024-07-29:
May the difference be 32-bit (Qt5) or 64-bit (Qt6) releases? On our [download page](https://librepcb.org/download/) we currently provide 64-bit Windows releases as BETA, maybe the bug doesn't exist in this build?

xotho, 2024-07-29:
The version that do not expose the issue already use v6.6.2 QT and it's a night build BETA.

[image]

LibrePCB Version: 1.1.1-unstable
Git Revision:     04485b3b5
Build Date:       2024-07-22T16:44:17
Build Author:     LibrePCB CI
Qt Version:       6.6.2 (built against 6.6.2)
CPU Architecture: x86_64
Operating System: Windows 11 Version 23H2
Platform Plugin:  windows
TLS Library:      OpenSSL 3.0.12 24 Oct 2023
OCC Library:      OCCT 7.7.2

// -------------------------------------------------------------------------------------------------------------------

The night build BETA that expose this issue uses v6.6.2 QT too, and it's : 

[image]

LibrePCB Version: 1.1.1-unstable
Git Revision:     38facb33b
Build Date:       2024-07-25T11:15:46
Build Author:     LibrePCB CI
Qt Version:       6.6.2 (built against 6.6.2)
CPU Architecture: x86_64
Operating System: Windows 11 Version 23H2
Platform Plugin:  windows
TLS Library:      OpenSSL 3.0.12 24 Oct 2023
OCC Library:      OCCT 7.7.2

ubruhin, 2024-07-29:
`38facb33b` is not on `master` branch - please do NOT use non-master branches, they are incomplete and dangerous! And for real work, please don't even use nightly builds at all, only use stable releases. It's a lot of wastes time for us to investigate issues on development builds, which hinders us to work on new features etc. If this happens more often, I'll think about not providing nightly builds at all anymore.

Closing as the issue is not relevant.

closed 2024-07-29 by ubruhin