LibrePCB/LibrePCB #1434: Wrong "clearance plane <-> board outline" DRC warning

dbrgn, 2024-09-07:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 1.1.0
Git Revision:     18a3d4589
Build Date:       2024-08-26T14:06:04
Qt Version:       6.7.2 (built against 6.7.2)
CPU Architecture: x86_64
Operating System: Arch Linux
Platform Plugin:  wayland
TLS Library:      OpenSSL 3.3.2 3 Sep 2024
OCC Library:      OCCT 7.7.2
```

##### SUMMARY

I get a "Clearance plane <-> board outline < 0.3 mm" warning, even though my plane clearance is set to 0.3 mm:

[image]

My design rules are set to 0.3 mm as well:

[image]

If I change the "Min. Clearance" in the plane properties to 0.31 mm, the DRC warning disappears. This leads me to think that it must be some kind of rounding error.

##### STEPS TO REPRODUCE

Here's the project: https://github.com/dbrgn/chicken-coop/tree/9e1b2b806a38a8aa39e1354edf54af91f443f064/ir-led-panel

##### EXPECTED RESULTS

No DRC warnings.

##### ACTUAL RESULTS

Two DRC warnings, one for each top/bottom plane.

ubruhin, 2024-09-07:
Yeah the clearance check is never an exact thing, I do allow a small amount of tolerance (e.g. 0.2999mm clearance should not raise the warning) to avoid getting false-positives but it seems this is not enough tolerance. Generally I don't want to allow too much tolerance because clearance is an important metric (and could possible raise a problem later in production).

I already was thinking about not checking actual clearances of planes at all, and only rely on the *configured* value. But this sounds a bit risky to me (e.g. if there is a bug in the plane calculation, we still have the DRC as a safety net to catch the problem). Maybe it would be possible to combine both mechanisms - check both the configured value, and if it conforms to the design rules, also check the actual clearances with a bigger tolerance :thinking:

dbrgn, 2024-09-07:
> Maybe it would be possible to combine both mechanisms - check both the configured value, and if it conforms to the design rules, also check the actual clearances with a bigger tolerance 🤔

That sounds like a good idea to me!

ubruhin, 2024-10-05:
...after thinking about it, probably such a cleaner solution doesn't work. In the end, the configured clearance value on a plane is not enough "evidence" that it violates any clearance. Consider a plane which gets fully filled (no traces or board outlines in its area) - in such a case, the configured clearance value has no effect - emitting a DRC warning about a too small clearance would be a false-positive then, so we end up with the same problem.

Maybe it's good enough to just increase the allowed tolerance from 5um to 10um (i.e. suppress clearance warnings if the violation is <=10um). I did this in #1442 and it fixes the false-positive warning in your provided project.

PR #1442 (2024-10-09): DRC: Fix false-positive board<->copper clearance errors
Fixes #1434
closed 2024-10-09 by ubruhin