LibrePCB/LibrePCB #1473: DRC should not check/complain for devices

eto-, 2024-12-05:
##### SUMMARY

There are several application for which a device has 2 pin in short in the package. It is a task of the designer verifying that this makes sense. I do not think that the DRC should signal an error if two lines are shorted within a device/package.

One typical occasion is to have a line changing its name in the schematics: for examples several lines coming from a connector with logical names are joint in the schematics. Having a single name may not respect the documentation of the mating connector. I typically use a "dot" to join these lines.
[image]
In the board it is implemented as two overlapping small pads. However the DRC generates several errors:
[image]
(I used on small nets on porpoise to avoid overlapping the lines here)

##### SOLUTION

ubruhin, 2024-12-05:
I totally agree, this is a typical use-case named "net-tie".

I think the main question is whether this is just a matter of the DRC ignoring in-package short circuits, or if there are more things to consider. For example too small clearances in packages can really be dangerous so one might not just suppress them - maybe packages need a property to specify them as net-tie (so only those are ignored by the DRC), or something similar.

eto-, 2024-12-06:
A simple implementation could be to add a flag to the package: if the flag is active the DRC ignores connections within the boundaries of the package outline.