LibrePCB/LibrePCB #1474: crash on start on mac os 11.7.10

stanelie, 2024-12-08:
The latest release 1.2.0 crashes on launch on my iMac running 11.7.10, the latest version this iMac can run.

Wanting to move away from EasyEDA, I found Horizon EDA (PC only), KiCad and LibrePCB. Wanting to try them all, I started with LibrePCBé. As I wanted to work on my older iMac that has a big screen, I find this is a turnoff, made worse by the fact that it looks like LibrePCB still works on even older versions of Mac OS (10.14) if I were to compile it myself.

Why not release a binary that works all the way down to 10.14, and be inclusive of the older OS?

ubruhin, 2024-12-08:
Hi,
Glad to hear you are trying LibrePCB :)

The thing is, our binary releases are built on CI (Azure Pipelines) for various reasons:

- To ensure correctness (e.g. not messed up by wrong manual building)
- To ensure reproducible results since everything is automated and in a well defined environment
- To save a lot of time (manually creating releases takes a lot of time)
- To increase transparency (everyone can see publicly *how* the releases are built)
- To avoid the requirement for me to have every release environment locally (for example, I do not own an Intel Mac, and building in a emulated environment is cumbersome and veeeery slow)
- We need to build LibrePCB on CI anyway for development purposes, so it's a small step to also use it for releasing

However, we can only build on systems provided by Azure Pipelines. Unfortunately they dropped support for MacOS 11 & 12, so we can no longer build on these systems, although LibrePCB would still be compatible with these systems.

So in the end we don't really have a way to provide builds for older MacOS versions. I'd love to do so, but tech companies make our life really hard to do so. For Linux it's no problem since we can build in Docker images, but Docker doesn't exist for Mac.

I'm aware there are people using older Mac versions, that's why I added the "can build from sources" note on our download page.

stanelie, 2024-12-08:
I understand, thanks for the detailed explanation, I appreciate it.

Cheers!

ubruhin, 2024-12-08:
Closing this as there's not much we could do.

Hope you'll enjoy LibrePCB built from sources :slightly_smiling_face:

closed 2024-12-08 by ubruhin

chriskuku, 2025-03-26:
I started trying to build from sources and it ain't no fun either.

```
 cd build
$ cmake ..
-- Building LibrePCB with Qt 6.2.3
-- Using vendored DelaunayTriangulation
-- Using vendored Dxflib
-- Using vendored FontoBeneQt
-- Building fontobene-qt with Qt 6.2.3
-- Using vendored MuParser
CMake Deprecation Warning at libs/muparser/CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 3.10 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

-- Using system OpenCASCADE (OCCT 7.8.0)
CMake Error at cmake/FindOpenCascade.cmake:22 (message):
  Cannot use OpenCascade 7.8.0 because it contains a critical bug.  Please
  use a different version (or if this is not possible, try OCE instead of
  OCCT).
Call Stack (most recent call first):
  CMakeLists.txt:164 (find_package)

-- Configuring incomplete, errors occurred!
```