LibrePCB/LibrePCB #1476: Almost no text in schematic window

vicencb, 2024-12-08:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 1.2.0
Git Revision:     f079fa97b
Build Date:       2024-12-08T20:52:56
Qt Version:       6.8.0 (built against 6.8.0)
CPU Architecture: x86_64
Operating System: Alpine Linux v3.21
Platform Plugin:  wayland
TLS Library:      OpenSSL 3.3.2 3 Sep 2024
OCC Library:      OCCT 7.8.1
```

##### SUMMARY
The only text shown in the schematic window of LibrePCB are net names.
[image]

##### STEPS TO REPRODUCE
Build and run LibrePCB on Alpine Linux, then open one of the example projects.

##### EXPECTED RESULTS
The `Export PDF ...` and `Export Image ...` work as expected, showing all text.
[image]

##### ADDITIONAL INFORMATION
LibrePCB was build from the original unmodified sources `librepcb-1.2.0-source.zip` with `CC=clang CXX=clang++ cmake -B build . -DQT_MAJOR_VERSION=6`

LibrePCB 1.1.0 with QT5 works fine on Alpine Linux.

On Alpine Linux it is not possible to run the pre-built versions `librepcb-1.2.0-linux-x86_64.AppImage` nor `librepcb-1.2.0-linux-x86_64.tar.gz` because of the different libc.

The full test suite passed, except for one single test:
```
$ build/tests/unittests/librepcb-unittests --gtest_filter='-EagleProjectImportTest.testNodino'
Note: Google Test filter = -EagleProjectImportTest.testNodino
[==========] Running 2269 tests from 130 test suites.
...
[  PASSED  ] 2269 tests.

$ build/tests/unittests/librepcb-unittests --gtest_filter='EagleProjectImportTest.testNodino'
Note: Google Test filter = EagleProjectImportTest.testNodino
[==========] Running 1 test from 1 test suite.
[----------] Global test environment set-up.
[----------] 1 test from EagleProjectImportTest
[ RUN      ] EagleProjectImportTest.testNodino
Segmentation fault
```

ubruhin, 2024-12-08:
Hi @vicencb,

Thanks for the detailed bug report! We just discussed this issue in our dev chat a few days ago and figured out it is a bug in Qt 6.8.0: https://bugreports.qt.io/browse/QTBUG-129914

With Qt 6.8.1 the bug is gone, so I hope it will be available in Alpine very soon. Or any other Qt version >= 6.2 would be fine too.

> The full test suite passed, except for one single test:

This however is something new to me and should be investigated :thinking:

ubruhin, 2024-12-08:
Btw:

> Platform Plugin:  wayland

Be aware that there are some known issues with wayland: #1249

ubruhin, 2024-12-09:
I tried to investigate the unittest crash, but unfortunately without success. I could reproduce it with Alpine 3.21 and 3.20 (in Docker container) but it happens on no other system we test on.

As I can't debug with QtCreator through Docker, it's very difficult to investigate. I already spent quite some time and need to stop at this point. I'd appreciate any help on this - if you have QtCreator available on Alpine, it should be much easier to just start the debugger from there.

As the original problem (missing text) is fixed in Qt6.8.1, I'll close this issue. But feel free to respond here if you'd be willing to investigate the unittest crash.

closed 2024-12-09 by ubruhin

vicencb, 2024-12-10:
Regarding the crash:

`librepcb::BoardNetSegmentSplitter::split()` calls `librepcb::BoardNetSegmentSplitter::findConnectedLinesAndPoints` with `availableTraces.first()->getStartPoint()` as first argument `anchor`.

Then, `findConnectedLinesAndPoints` mutates the `availableTraces` via the `remove` method in a loop.

The first iteration of the loop is fine, because the reference `anchor` is used before changing the memory it points to, but the second iteration ...

ubruhin, 2024-12-10:
Huh that's really really not good :scream: Strange this code has been untouched for more than 4 years and never caused a problem on CI (on 3 different systems!) and can't remember anyone ever reported a crash related to this. I wonder if we should add a CI job for Alpine some day...

Implemented a fix in #1478 and verified on Alpine the unit test no longer segfaults. Probably on Alpine you should build LibrePCB with this patch included as the affected code is also used in the board & schematic editors when pasting or deleting elements.

Thanks for your help!