LibrePCB/LibrePCB #1480: Unable to download libraries: "QSslSocket::connectToHostEncrypted: TLS initialization failed" with released AppImage

eduardosm, 2024-12-14:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 1.2.0
Git Revision:     f079fa97b
Build Date:       2024-12-01T10:28:34
Build Author:     LibrePCB CI
Qt Version:       6.6.2 (built against 6.6.2)
CPU Architecture: x86_64
Operating System: Fedora Linux 41 (KDE Plasma)
Platform Plugin:  xcb
TLS Library:      
OCC Library:      OCCT 7.7.2
```

##### SUMMARY

It is not possible to download libraries.

##### STEPS TO REPRODUCE

1. Launch LibrePCB.
2. Open the library manager.

##### EXPECTED RESULTS

The "Install Libraries" tab should show a list of libraries available to install.

##### ACTUAL RESULTS

The list is empty.

##### ADDITIONAL INFORMATION

I'm using the AppImage version.

Console output:
```
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
[  INFO   ] LibrePCB 1.2.0 (f079fa97b)
[  INFO   ] Qt version: 6.6.2 (compiled against 6.6.2)
[  INFO   ] Resources directory: "/home/eduardosm/librepcb-1.2.0-linux-x86_64/share/librepcb"
[  INFO   ] Application settings: "/home/eduardosm/.config/LibrePCB/LibrePCB.ini"
[  INFO   ] Cache directory: "/home/eduardosm/.cache/LibrePCB/LibrePCB"
[DEBUG-MSG] Network access manager thread started.
[DEBUG-MSG] Recently used workspace: "/home/eduardosm/LibrePCB-Workspace"
[DEBUG-MSG] Detected dark theme based on window background color #2a2e32.
[DEBUG-MSG] Open workspace data directory "/home/eduardosm/LibrePCB-Workspace/data"...
[ WARNING ] Override stale lock on directory "/home/eduardosm/LibrePCB-Workspace/data"...
[DEBUG-MSG] Load workspace settings...
[DEBUG-MSG] Successfully loaded workspace settings.
[DEBUG-MSG] Load workspace library database...
[DEBUG-MSG] Successfully loaded workspace library database.
[DEBUG-MSG] Successfully opened workspace.
[DEBUG-MSG] Workspace library scanner thread started.
[DEBUG-MSG] Start workspace library scan in worker thread...
[DEBUG-MSG] Workspace libraries indexed: 0 libraries in 0 ms.
[DEBUG-MSG] Workspace library scan succeeded: 0 elements in 0 ms.
[ WARNING ] The backend "cert-only" does not support QSslKey
[ WARNING ] Active TLS backend does not support key creation
[ WARNING ] The backend "cert-only" does not support QSslKey
[ WARNING ] Active TLS backend does not support key creation
[ WARNING ] The backend "cert-only" does not support QSslKey
[ WARNING ] Active TLS backend does not support key creation
[ WARNING ] The backend "cert-only" does not support QSslKey
[ WARNING ] Active TLS backend does not support key creation
[ WARNING ] The backend "cert-only" does not support QSslKey
[ WARNING ] Active TLS backend does not support key creation
[ WARNING ] The backend "cert-only" does not support QSslKey
[ WARNING ] Active TLS backend does not support key creation
[ WARNING ] The backend "cert-only" does not support QSslSocket
[ WARNING ] The backend named "cert-only" does not support TLS
[ WARNING ] QSslSocket::connectToHostEncrypted: TLS initialization failed
[CRITICAL ] Request failed: "https://api.librepcb.org/api/v1/libraries/v1"
[CRITICAL ] Network error: "TLS initialization failed (99)"
```

ubruhin, 2024-12-14:
Arghh, seems there's something broken with libssl in the Linux binaries :-/ Very strange the AppImage works on other distributions.

As a workaround it *might* be possible to use the [*.tar.gz](https://download.librepcb.org/releases/1.2.0/librepcb-1.2.0-linux-x86_64.tar.gz) and delete all files in `lib/` with "ssl" or "crypto" in their name, to use libssl from your system instead of the bundled one.

ubruhin, 2025-03-18:
That was a tricky one... after many hours of investigation I think I could fix it in #1495 

Here are nightly builds which during my testing now work properly also on Fedora: https://download.librepcb.org/nightly_builds/update-ci/

@eduardosm any feedback would be appreciated if these nightly builds now work fine on your system.

eduardosm, 2025-03-18:
Nightly's AppImage now works out of the box for me. Thanks!

closed 2025-03-23 by ubruhin