LibrePCB/LibrePCB #1486: GenCAD export (production data)

awatterott, 2025-01-20:
##### SUMMARY
The generation of production data in GenCAD format would be a good addition. It is needed for solder paste printers like the Mycronic MY700.

##### SOLUTION

ubruhin, 2025-01-21:
Hmm, never heard of that 🙈 

If it goes to a machine, isn't this more a job for a CAM software which imports the solder paste layer from Gerber files? Or is GenCAD just a representation of the PCB image, fully independent of any machine parameters?

If you have any further information, it would be appreciated. Also we'd need things like that:

- GenCad file format specification and/or example files
- (free) software to allow verifying the exported files

awatterott, 2025-06-26:
The newest Mycronic CAM software supports ODB++. So this can be closed.

closed 2025-06-26 by awatterott