LibrePCB/LibrePCB #1494: Tracking issue: New UI for LibrePCB 2.0

ubruhin, 2025-03-16:
_(this is a tracking issue about the progress if the new UI currently in development and will be updated regularly)_

# Goals

The goals of the new UI, compared to the current UI:

- Less windows, suitable for both single-monitor and multi-monitor setups
  - To be implemented with generic windows containing the different editors as tabs
- Less modal dialogs & message boxes to avoid distracting the user
- Efficient usage of screen space (e.g. not wasting lots of space with half-empty toolbars)
- Dynamic, context-aware content (e.g. only show relevant information depending on state)
- Responsive design (e.g. make content dependent on window size)
- Intuitive design, with tooltips, hints etc. helping and guiding the user
- Generally more modern look, with animations where appropriate
- More efficient development workflow
- Slowly moving away from C++ & Qt towards Rust

See also the blog post [NGI0 Grant for LibrePCB 2.0](https://librepcb.org/blog/2024-10-17_roadmap_2.0/).

# Tasks in Other Repositories

- [ ] https://github.com/LibrePCB/docker-librepcb-dev/pull/57

# Tasks in This Repository

- [ ] Update CI to new Docker images
- [ ] Replace ControlPanel with new Slint UI
- [ ] Integrate AboutDialog into new Slint UI
- [ ] Integrate LibraryManager into new Slint UI
- [ ] Integrate SchematicEditor & BoardEditor into new Slint UI
- [ ] Integrate OrderPcbDialog into new Slint UI
- [ ] Integrate OutputJobsDialog into new Slint UI
- [ ] Integrate LibraryEditor into new Slint UI
- [ ] Integrate WorkspaceSettingsDialog into new Slint UI

# Issues to be Solved or Improved

Note that not all issues are expected to be fully solved in LibrePCB 2.0, some might be completed in a later release.

## Solved By Design / Not Relevant Anymore

- [ ] https://github.com/LibrePCB/LibrePCB/issues/36
- [ ] https://github.com/LibrePCB/LibrePCB/issues/53
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1050

## Problems of Current UI, Not Relevant Anymore or Easy to Fix

- [ ] https://github.com/LibrePCB/LibrePCB/issues/903
- [ ] https://github.com/LibrePCB/LibrePCB/issues/942
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1232
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1371
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1372
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1417

## To Be Considered / Reviewed

- [ ] https://github.com/LibrePCB/LibrePCB/issues/359
- [ ] https://github.com/LibrePCB/LibrePCB/issues/726
- [ ] https://github.com/LibrePCB/LibrePCB/issues/815
- [ ] https://github.com/LibrePCB/LibrePCB/issues/915
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1053
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1231
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1258
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1362
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1367
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1424

## To be Implemented For Sure

- [ ] https://github.com/LibrePCB/LibrePCB/issues/1051

## To be Implemented Sooner or Later

- [ ] https://github.com/LibrePCB/LibrePCB/issues/646
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1045
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1254
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1332
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1354
- [ ] https://github.com/LibrePCB/LibrePCB/issues/1370

# Related External Issues

- [ ] https://github.com/slint-ui/slint/issues/3715

melak, 2025-03-24:
How about a command input field, much the same way it was done in Eagle (and in several other CAD-type things, I suppose)?
A halfway house (i.e. the way QCAD does it) is better than nothing, but a full one would be best of course (to be honest, this is the thing I'm missing the most).

ubruhin, 2025-03-24:
> How about a command input field, much the same way it was done in Eagle (and in several other CAD-type things, I suppose)?

Yeah that was already requested some time ago, I think there's an issue open for that. But this goes much further than just a UI feature as it would require kind of a script interpreter - a huge amount of work.

Or maybe you mean it simpler: I know a command input field from Qt Creator (Ctrl+K) and VS Code (F1) which basically provide all the actions from the menu (and possibly more) in a keyboard-friendly way (e.g. press F1, type "Ope" and you'll get the suggestion "Open Project"). That's something I'd like to implement but I'm not sure if in the 2.0 release already.