LibrePCB/LibrePCB #1511: Improve Schematic Readibility based on ZOOM dimension

xotho, 2025-04-11:
I am facing a graphic behavior into the SCHEMATIC EDITOR compared to other application.

When zooming out NET visual reading is reduced or even lost.

Here I show an example about Libre and another application:

[image]

[image]

As you can see the readability is close to ZERO in Libre PCB but in other application is still 'visible'.

I know this example is exeding readibility limits, but even at lower ZOOM reduction visual info are lost.

can readability be improved ? by changing (more bright green) color ?  by changing NET dimension reduction % when zooming out ? A mix of the two ?

Just a suggestion because this reflects really a lot in schematic readability.

Here a sample with the same ZOOM level:

[image]

take care,
X!

ubruhin, 2025-04-11:
Maybe we could somehow ensure a minimum line width of 1px even if the zoom level would require less than 1px.

xotho, 2025-04-11:
Agree 100%, 
maybe also an anti-alias enabled setting of the graphic primitives can make the visibility worse too if enabled ?

take care,
X!