LibrePCB/LibrePCB #1514: LCSC / EasyEDA Component, Footprint, 3D-Model Import

hb9gvd, 2025-04-21:
As JLCPCB has LCSC / EasyEDA as _Basic Parts_ for their inexpensice PCB assembly, an import function to _My LibrePCB part library_ or another way of interoperabilty with LCSC parts will improve acceptance of LibrePCB in the maker community.

ubruhin, 2025-04-22:
Hmm yeah sounds like a good idea. I just found https://github.com/uPesy/easyeda2kicad.py which could be helpful (as a reference) to implement such a feature in LibrePCB.

z3ugma, 2025-05-05:
I have also long sought this, and there's maybe a new interaction modality. The current concept of a library implies that you will download components. 

There's a good analog with how Partstack shows live pricing information to the BOM - it's more like a live search for parts because the library updates often?

z3ugma, 2025-05-05:
Further idea: since the library is so huge, along the likes of the DigiKey catalog or the Mouser catalog, it's more like a "parts search" window, that links out to the datasheet and to pictures of the part. 

[image]

With an import dialog, maybe it's a list of LCSC ID#s or a text-entry input field in which you enter the LCSC ID, and then show a confirmation dialog of the part, before proceeding with import or update to the local library

ubruhin, 2025-05-06:
Yes a feature like this will be very different to the Eagle & KiCad library importers since this library is huge and we can only import parts "on demand", i.e. one by one - not a whole library at once.

Note that currently I'm very busy with #1494 so at the moment I really don't have time to take care about this. But as said, generally I like the idea.

z3ugma, 2025-05-06:
Completely understood. If you're open to inputs or mockups on UI for it, I've got some ideas, but there don't appear to be Nightly builds for macOS ARM architectures to install and try out #1494 and it might take me a bit to get the build toolchain working on my ARM mac. 

_Are_ there nightly macOS ARM builds?

ubruhin, 2025-05-06:
For ARM there are no nighty builds unfortunately. But I guess just for trying out it should be good enough to run the x86 build with Rosetta? Anyway, building LibrePCB from sources is really not hard on macOS, just follow our README.md. And if you have any problems, join us in our Matrix space (see https://librepcb.org/help/).

Of course I'd be interested to hear your ideas regarding UI. I do have some ideas as well but there are also areas where I'm not sure yet. Note that currently I'm working on the `slint-project-editor` branch which is much further than the master branch.

z3ugma, 2025-06-01:
Copying in some discussion from Matrix chat:

I'm starting work on an EasyEDA > LibrePCB library UI
I'll use PySide6 and Qt6 for the interface and Python for the EasyEDA to LibrePCB s-expressions primitives for library conversion. Basing it on https://github.com/uPesy/easyeda2kicad.py. It will create a local library on the user's computer in the directory of their choosing and add "their" EasyEDA components searched in the UI to the local library

philosophically I'm interested in a lot of the commentary I see in the forums and in GitHub issues: Library conversion is hard, pros have to review footprints, converters and different EDAs follow different conventions. 

I want to incorporate a review workflow into the tool. 
Actually, your addition of Footprint image inclusion in a recent LibrePCB release is a great foundation for this

Dream project goal is that a Web based library > search > copy to local > review footprints/schematics can be "the Library" workflow that solves a lot of the commentary about not enough components being available in the standard libraries 

Alright 👍️ I think for long term it would be nice to have this feature integrated in LibrePCB, but I'm not sure yet if the conversion should be done locally like Eagle/KiCad importers (then it should be written in Rust or C++) or if it should be done on our server and integrated into the application through the API because it relies on API access of EasyEDA which could theoretically break at any time.

Ok, I'll keep doing an independent workflow tool but try to keep it portable enough that I could pitch you on lifting/porting/incorporating it after the new UI rewrite / 2.0 are done

>Urban
>Yes maybe, but PartStack is also integrated through our API. The LibrePCB source code does not contain anything Partstack specific because that would be too high risk of breakage. Also it allows to switch to another partner at any time when needed.

FWIW I do like the server route - where you can have part search that is vendor-agnostic in the server API. Have a "LibrePCB parts search" that uses the API endpoint listed in Settings (so that if someone really wanted to roll another compatible API server they could) and extend the existing "parts" https://github.com/LibrePCB/librepcb-api-server/blob/master/app.py#L261 call to also be able to send footprints/symbols. 

Then, the workflow UI tool is focused on search > results review > copy part into local library > review symbols for accuracy 
An extension of this could be: after an individual reviews the footprint and "approves" it, then keep a count of "approved" footprints that gets sent back to the server for a "confidence score" of how good the part is in the server library.