LibrePCB/LibrePCB #1515: CMakeLists.txt compatibility error

Rioboyva2554, 2025-04-25:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 1.3.0 
Git Revision: latest 
Qt Version: 6
Operating System: I use arch btw
...
```

##### SUMMARY

When I compile with Cmake it throws this error about the minimum compatibility version specified in libs/type_safe/CMakeLists.txt to be under 3.5. I Know how to fix this issue, But I am letting you know 

##### STEPS TO REPRODUCE

compile it

##### EXPECTED RESULTS

to compile

##### ACTUAL RESULTS

threw error

##### ADDITIONAL INFORMATION

``` -- Using vendored TypeSafe
CMake Error at libs/type_safe/CMakeLists.txt:5 (cmake_minimum_required):
  Compatibility with CMake < 3.5 has been removed from CMake.

  Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
  to tell CMake that the project requires at least <min> but has been updated
  to work with policies introduced by <max> or earlier.

  Or, add -DCMAKE_POLICY_VERSION_MINIMUM=3.5 to try configuring anyway. ```

ubruhin, 2025-04-25:
Thanks for reporting. This has been fixed in #1506, unfortunately *after* the 1.3.0 release :-/

closed 2025-04-25 by ubruhin

Rioboyva2554, 2025-04-25:
Sorry I didn't see it