LibrePCB/LibrePCB #1530: librepcb-cli / additional switches and machine-readable --check messages

z3ugma, 2025-05-28:
##### SUMMARY
`librepcb-cli` library checks print human-friendly output. 
Request is to provide the output of these messages in a machine-readable way, possibly using the same S-expressions that are used when you ignore the message in the UI.

```
librepcb-cli open-library --all -v --check .
```
1. Add a switch to librepcb-cli that lets you check a specific library subtype: pkg, cmp, etc 
2. Add a switch to librepcb-cli that lets you specify a specific library element by UUID or filepath
2. Provide the output of the `--check` messages in a machine-readable way for automatically approving warnings

Note: when manually applying an approval, this is the sexpr that gets saved into the `package.lp` file 
```
 (approved small_pad_clearance
  (footprint b0d5a736-74be-41d6-a029-8cc741e4302b)
  (pad 67c82d52-6ec8-42f0-afbe-fd0367defd5e)
  (pad 6b1ea7ae-df5b-465d-833b-4dd8f8685afa)
 )
```

z3ugma, 2025-06-02:
Finding the need for other switches here too: 

 - export_image for packages and schematics for visual side-by-side inspection in manual test suites for library conversion and import. This is already supported in the CLI for _boards_ but not for _library elements_.

ubruhin, 2025-06-04:
I'm trying to understand the actual use-cases to decide which features are reasonable to implement. Let me share my thoughts about your described use-cases.

> Provide the output of the --check messages in a machine-readable way for automatically approving warnings

Do we really need this if #1529 is implemented? Since it should stop LibrePCB from emitting those warnings, honestly I don't see a need anymore for mass-approving messages. At least I think it's a bad idea to mass-approve messages you didn't review manually.

> export_image for packages and schematics for visual side-by-side inspection in manual test suites for library conversion and import. This is already supported in the CLI for boards but not for library elements.

> Add a switch to librepcb-cli that lets you check a specific library subtype: pkg, cmp, etc

What if the flag for exporting images are actually two flags, `--export-symbols` and `--export-footprints` so everything can be exported in a single call to `librepcb-cli` and no filtering on subtype is needed anymore? The passed filepaths would then be patterns which are substituted by the element's name or UUID, e.g.

    librepcb-cli open-library --all --export-footprints "/export/to/{{PACKAGE_UUID}}_{{FOOTPRINT_NAME}}.png"

> Add a switch to librepcb-cli that lets you specify a specific library element by UUID or filepath

Not sure if this would also be related to the use-cases described above, but at least for our `librepab-parts-generator` this could be a useful feature so I agree to implement it. But I'd make this individual commands (e.g. `librepcb-cli open-symbol [...]`) because library elements are generally independent of libraries so it should be possible to open them without having a library. One could think about having a generic `open-library-element` rather than up to 6 individual commands, but I think individual commands would be cleaner since the flags will depend on the element type (e.g. `--export-image` makes no sense on a component).

I'm just trying to understand what we really should implement. I want to avoid implementing features which just add more complexity for things nobody actually needs. Also for very specific needs (which I'd even consider the visual side-by-side comparison) actually a Python package would make more sense as it would be much more flexible than a CLI. But since we don't have a Python package yet I agree *some* functionality could be integrated into the CLI for now.

ubruhin, 2025-06-04:
Alright - so I think it makes sense to focus on the `open-symbol` etc. commands for now as this should solve quite a few issues without introducing too much complexity (and also isn't that complicated to implement). I updated the issue title accordingly. And I'm open to accept contributions for this.

Commands to be supported:

- `--export` to export graphics of symbols & footprints (I think it's nicer than `--export-image` because PDF is strictly not an image but internally both are handled the same way). File format to be auto-detected from file extension.
- `--check`
- `--minify-step` for packages
- `--save`
- `--strict`

Not all commands and all library element types need to be implemented at once, even a contribution for a subset would already be nice.

For the `--export-checks` I'm not 100% convinced yet. When running checks on a particular library element it might be enough to just capture the console output and show it as-is to the user of a mass-migration tool?

And btw @z3ugma as you asked for ARM MacOS nightly builds in another issue: We do have them now (see https://download.librepcb.org/nightly_builds/master/) but they require at least MacOS 14 and I wasn't able yet to actually test them.

z3ugma, 2025-06-18:
TODO: 
 - [ ] add support for `open-package` and `open-symbol`
 - [ ] add support for `open-component` and `open-device`
 - [ ] update `librepcb-doc/modules/cli/pages` to have documentation for the new CLI commands

Out of scope for me for now:
 - [ ] come back and add support for `--check` and `--save` and `--strict` to all 4 library element types

z3ugma, 2025-06-23:
@ubruhin sometimes there are messages that are not uniquely identifiable from the output of the CLI tool. when a message is Approved, it gets saved in s-expressions with more specific output via its UUID in the .lp file 

```
(approved thin_line
  (footprint 39e1b05b-30bd-4c64-a6c9-a1b67d9eb207)
  (polygon 1120576e-d63d-4dfb-b028-e8b6c80cd23f)
 )
```

[image]

if I want to reference the messages but not their approvals then I need some way to be able to distinguish among messages with similar content produced for different polygons...

```
      "message": "Minimum width of 'Top Legend' in 'default'",
      "severity": "WARNING"
    },
    {
      "message": "Minimum width of 'Top Legend' in 'default'",
      "severity": "WARNING"
    },
   
```