LibrePCB/LibrePCB #1593: Build failure on FreeBSD (14)

melak, 2025-08-31:
The environment is FreeBSD 14.3, where I guess the point is non-base-OS software is installed in /usr/local where /usr/local is *not* in the *default* compiler/linker search path (but build systems and package things are aware of this, and things tend to work just fine).

If I build LibrePCB `slint` by itself, things are apparently fine:

```
https://github.com/LibrePCB/slint.git@librepcb-patches

freebsd:~/w/librepcb/slint(librepcb-patches)$ git describe HEAD --always
c0d73450e
  
freebsd:~/w/librepcb/slint(librepcb-patches)$ cmake .
-- The CXX compiler identification is Clang 19.1.7
...
freebsd:~/w/librepcb/slint(librepcb-patches)*$ make
Built target cargo-prebuild_librepcbslint
...
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 2m 30s
Copying byproducts `slint-compiler` to /home/ice/w/librepcb/slint/api/cpp
Built target _cargo-build_slint-compiler
Built target cargo-build_slint-compiler
Built target Slint
```

If I build LibrePCB itself, `slint` fails to build by not finding things (that are definiely there, of course):

```
freebsd:~/w/librepcb/LibrePCB(master)$ git describe HEAD --always
1.3.0-288-gfc48f3285

freebsd:~/w/librepcb/LibrePCB(master)$ cmake .
-- The CXX compiler identification is Clang 19.1.7
...
freebsd:~/w/librepcb/LibrePCB(master)*$ make
[  0%] Built target cargo-prebuild_librepcbslint 
... 
   Compiling itertools v0.14.0
warning: i-slint-backend-qt@1.13.0: In file included from qt_accessible.rs:266:
warning: i-slint-backend-qt@1.13.0: In file included from /usr/local/include/qt6/QtWidgets/QtWidgets:3:
warning: i-slint-backend-qt@1.13.0: In file included from /usr/local/include/qt6/QtWidgets/QtWidgetsDepends:4:
warning: i-slint-backend-qt@1.13.0: In file included from /usr/local/include/qt6/QtGui/QtGui:61:
warning: i-slint-backend-qt@1.13.0: /usr/local/include/qt6/QtGui/qopengl.h:115:13: fatal error: 'GL/gl.h' file not found
warning: i-slint-backend-qt@1.13.0:   115 | #   include <GL/gl.h>
warning: i-slint-backend-qt@1.13.0:       |             ^~~~~~~~~
warning: i-slint-backend-qt@1.13.0: 1 error generated.
error: failed to run custom build command for `i-slint-backend-qt v1.13.0 (/home/ice/w/librepcb/LibrePCB/libs/slint/internal/backends/qt)`

Caused by:
  process didn't exit successfully: `/home/ice/w/librepcb/LibrePCB/./cargo/build/release/build/i-slint-backend-qt-1467e49596b96cf9/build-script-build` (exit status: 1)
  --- stdout
  cargo:rustc-check-cfg=cfg(no_qt)
  cargo:rerun-if-env-changed=SLINT_NO_QT
  OUT_DIR = Some(/home/ice/w/librepcb/LibrePCB/./cargo/build/x86_64-unknown-freebsd/release/build/i-slint-backend-qt-f55876343c819a55/out)
  TARGET = Some(x86_64-unknown-freebsd)
  OPT_LEVEL = Some(3)
  HOST = Some(x86_64-unknown-freebsd)
  cargo:rerun-if-env-changed=CXX_x86_64-unknown-freebsd
  CXX_x86_64-unknown-freebsd = None
  cargo:rerun-if-env-changed=CXX_x86_64_unknown_freebsd
  CXX_x86_64_unknown_freebsd = Some(/usr/bin/c++)
  RUSTC_WRAPPER = None
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  DEBUG = Some(false)
  cargo:rerun-if-env-changed=CXXFLAGS_x86_64-unknown-freebsd
  CXXFLAGS_x86_64-unknown-freebsd = None
  cargo:rerun-if-env-changed=CXXFLAGS_x86_64_unknown_freebsd
  CXXFLAGS_x86_64_unknown_freebsd = None
  cargo:rerun-if-env-changed=HOST_CXXFLAGS
  HOST_CXXFLAGS = None
  cargo:rerun-if-env-changed=CXXFLAGS
  CXXFLAGS = None
  OUT_DIR = Some(/home/ice/w/librepcb/LibrePCB/./cargo/build/x86_64-unknown-freebsd/release/build/i-slint-backend-qt-f55876343c819a55/out)
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  HOST = Some(x86_64-unknown-freebsd)
  cargo:rerun-if-env-changed=CXXFLAGS_x86_64-unknown-freebsd
  CXXFLAGS_x86_64-unknown-freebsd = None
  cargo:rerun-if-env-changed=CXXFLAGS_x86_64_unknown_freebsd
  CXXFLAGS_x86_64_unknown_freebsd = None
  cargo:rerun-if-env-changed=HOST_CXXFLAGS
  HOST_CXXFLAGS = None
  cargo:rerun-if-env-changed=CXXFLAGS
  CXXFLAGS = None
  OUT_DIR = Some(/home/ice/w/librepcb/LibrePCB/./cargo/build/x86_64-unknown-freebsd/release/build/i-slint-backend-qt-f55876343c819a55/out)
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS 
  CRATE_CC_NO_DEFAULTS = None
  HOST = Some(x86_64-unknown-freebsd)
  cargo:rerun-if-env-changed=CXXFLAGS_x86_64-unknown-freebsd
  CXXFLAGS_x86_64-unknown-freebsd = None
  cargo:rerun-if-env-changed=CXXFLAGS_x86_64_unknown_freebsd
  CXXFLAGS_x86_64_unknown_freebsd = None
  cargo:rerun-if-env-changed=HOST_CXXFLAGS
  HOST_CXXFLAGS = None
  cargo:rerun-if-env-changed=CXXFLAGS
  CXXFLAGS = None
  OUT_DIR = Some(/home/ice/w/librepcb/LibrePCB/./cargo/build/x86_64-unknown-freebsd/release/build/i-slint-backend-qt-f55876343c819a55/out)
  cargo:rerun-if-env-changed=CC_ENABLE_DEBUG_OUTPUT
  cargo:rerun-if-env-changed=CRATE_CC_NO_DEFAULTS
  CRATE_CC_NO_DEFAULTS = None
  HOST = Some(x86_64-unknown-freebsd)
  cargo:rerun-if-env-changed=CXXFLAGS_x86_64-unknown-freebsd
  CXXFLAGS_x86_64-unknown-freebsd = None
  cargo:rerun-if-env-changed=CXXFLAGS_x86_64_unknown_freebsd
  CXXFLAGS_x86_64_unknown_freebsd = None
  cargo:rerun-if-env-changed=HOST_CXXFLAGS
  HOST_CXXFLAGS = None
  cargo:rerun-if-env-changed=CXXFLAGS
  CXXFLAGS = None
  cargo:warning=In file included from qt_accessible.rs:266:
  cargo:warning=In file included from /usr/local/include/qt6/QtWidgets/QtWidgets:3:
  cargo:warning=In file included from /usr/local/include/qt6/QtWidgets/QtWidgetsDepends:4:
  cargo:warning=In file included from /usr/local/include/qt6/QtGui/QtGui:61:
  cargo:warning=/usr/local/include/qt6/QtGui/qopengl.h:115:13: fatal error: 'GL/gl.h' file not found
  cargo:warning=  115 | #   include <GL/gl.h>
  cargo:warning=      |             ^~~~~~~~~
  cargo:warning=1 error generated.

  --- stderr

  error occurred: ToolExecError: Command "/usr/bin/c++" "-O3" "-ffunction-sections" "-fdata-sections" "-fPIC" "-m64" "-stdlib=libc++" "--target=x86_64-unknown-freebsd" "-I" "/home/ice/w/librepcb/LibrePCB/libs/slint/internal/backends/qt" "-I" "/usr/local/include/qt6" "-Wall" "-Wextra" "-std=c++17" "-std=c++17" "-o" "/home/ice/w/librepcb/LibrePCB/./cargo/build/x86_64-unknown-freebsd/release/build/i-slint-backend-qt-f55876343c819a55/out/5c66a1c202f2662c-cpp_closures.o" "-c" "/home/ice/w/librepcb/LibrePCB/./cargo/build/x86_64-unknown-freebsd/release/build/i-slint-backend-qt-f55876343c819a55/out/rust_cpp/cpp_closures.cpp" with args c++ did not execute successfully (status code exit status: 1).
```

melak, 2025-08-31:
Building LibrePCB itself with Ninja is the same failure.

Building slint/api/cpp with either make or Ninja succeeds (apparently by default it turns `BACKEND_QT` on only on "Linux", so this starts to make sense now).

Building slint or slint/api/cpp with `SLINT_FEATURE_BACKEND_QT ON` (i.e. `cmake -DSLINT_FEATURE_BACKEND_QT=ON ...`) is the same failure, so it would seem like the culprit involves the detection or passing through or something of Qt c/ldflags probably to the Rust thingy?

The failure itself is coming from a Rust file, `qt_accessible.rs`, which has
```
cpp! {{
    #include <QtWidgets/QtWidgets>
```
in it. `QtWidgets/QtWidgets` (that is `/usr/local/include/qt6/QtWidgets/QtWidgets`) is apparently found correctly, and then a couple of layers, and then `QtGui/opengl.h` includes `<GL/gl.h>` which is not found, most likely because the Rust things didn't realise to pick `/usr/local/include` up (says without the slightest idea about Rust or Cargo).

melak, 2025-08-31:
```
freebsd:~/w/librepcb$ ls -l /usr/local/include/GL/gl.h 
-rw-r--r--  1 root wheel 80393 Nov  9  2022 /usr/local/include/GL/gl.h

freebsd:~/w/librepcb$ qmake6 -query
QT_SYSROOT:
QT_INSTALL_PREFIX:/usr/local
QT_INSTALL_ARCHDATA:/usr/local/lib/qt6
QT_INSTALL_DATA:/usr/local/share/qt6
QT_INSTALL_DOCS:/usr/local/share/doc/qt6
QT_INSTALL_HEADERS:/usr/local/include/qt6
QT_INSTALL_LIBS:/usr/local/lib/qt6
QT_INSTALL_LIBEXECS:/usr/local/libexec/qt6
QT_INSTALL_BINS:/usr/local/lib/qt6/bin
QT_INSTALL_TESTS:/usr/local/tests
QT_INSTALL_PLUGINS:/usr/local/lib/qt6/plugins
QT_INSTALL_QML:/usr/local/lib/qt6/qml
QT_INSTALL_TRANSLATIONS:/usr/local/share/qt6/translations
QT_INSTALL_CONFIGURATION:
QT_INSTALL_EXAMPLES:/usr/local/share/examples/qt6
QT_INSTALL_DEMOS:/usr/local/share/examples/qt6
QT_HOST_PREFIX:/usr/local
QT_HOST_DATA:/usr/local/lib/qt6
QT_HOST_BINS:/usr/local/lib/qt6/bin
QT_HOST_LIBEXECS:/usr/local/libexec/qt6
QT_HOST_LIBS:/usr/local/lib/qt6
QMAKE_SPEC:freebsd-clang
QMAKE_XSPEC:freebsd-clang
QMAKE_VERSION:3.1
QT_VERSION:6.9.2
```
I also have a `FindOpenGL.cmake` on both Linux and FreeBSD, apparently brought in by CMake itself, if that helps.

ubruhin, 2025-08-31:
Thanks :+1:

> I also have a FindOpenGL.cmake on both Linux and FreeBSD, apparently brought in by CMake itself, if that helps.

My suspicion is that it's not a CMake issue. If I understand correctly, [`qttypes/build.rs`] tries to find the include paths to compile the qttypes crate by itself, and is missing the include path for OpenGL. I'll ask the Slint devs tomorrow if my suspicion is correct or if they even know this problem already.

ubruhin, 2025-09-02:
> (and is now failing elsewhere but I guess you are aware of that).

LibrePCB is still compiling in my VM, but no error yet. What error do you get now? (compiling LibrePCB, not just Slint)

Only a compiler *warning* I saw, I'll fix that.

melak, 2025-09-02:
LibrePCB hasn't been updaed yet? Freshly `git clone --recursive`'ing LibrePCB comes back with
```
Submodule path 'libs/slint': checked out 'c0d73450e3cf510f47462d290484c3fb51ce4879'
```

```
pinky:/tmp/a/LibrePCB$ git describe HEAD
1.3.0-290-g1e6f8cfeb
pinky:/tmp/a/LibrePCB$ cat .git/modules/libs/slint/HEAD      
c0d73450e3cf510f47462d290484c3fb51ce4879
```

melak, 2025-09-02:
Oh, my bad. Let me check that.

melak, 2025-09-02:
Build completed successfully and the result runs, thank you very much ❤️

PR #1594 (2025-09-02): Update submodule "libs/slint"
Fixes #1593
closed 2025-09-02 by ubruhin