LibrePCB/LibrePCB #1599: Align/distribute objects in a matrix

ubruhin, 2025-09-15:
##### SUMMARY

The current alignment/distribution tool is not bad, but it lacks the possibility to place objects in a matrix. An example use case is to create an LED matrix by selecting all LED devices and specifying x/y pitch to place them all at once.

##### SOLUTION

tbd

JustLike1984, 2025-11-24:
> Wouldn't it be enough to have these as separate tools?,  mean, you can already rotate and flip parts in the board...

I don't know how the flip side is handled, but I'd assume that its part of the select tool, I'd want to re-use the underlying
function, so you can press the button, or press 'F' like you can in the select tool.

The rotation is of the parts in the matrix, or relative to the path they would be placed on.
I don't think there are any options for setting the absolute or relative rotation of multiple parts. 
besides editing the part parameters by hand for each one.

[image]
[image]

While there already are options to rotate a group of elements, rotating each individual element within a group is not
possible yet, relative or absolute. 

> Maybe the path should to be drawn first as a polygon in the board, and the tool allows you to select that polygon?

That is how i was already thinking of doing it.

> Any advanced selection modes (like entering names "D1, D7" or "D3..D8" or "D*" instead of rectangular mouse selection) can still be provided, but doesn't need to be part of the matrix tool

While i did consider using the place devices side pane, it wouldn't be enough as is.
Ideally it should be possible to select placed or unplaced parts. But adding an option to arrange selected parts from
the list could be quite handy. (although that would require getting multiple selections working)

The way i think this that you'd open it with an existing selection either from toolbar or context menu, or even place devices, or without one, and some way to select it.

you'd set the options you want, and click wherever you want the matrix to go. And until you exit out of the tool, or change the selection or deselect the position ( maybe an X button on the toolbar?), editing the options would apply to the matrix, so you can see what you are arranging. Before selecting the position, you'd get a preview.

ubruhin, 2025-11-24:
> I don't know how the flip side is handled, but I'd assume that its part of the select tool, I'd want to re-use the underlying
function, so you can press the button, or press 'F' like you can in the select tool.

I'm not against supporting flip, I just think it's not really the responsibility of the matrix tool because it is a general-purpose operation. My assumption is that users will place the parts on the desired board side anyway before invoking the matrix tool. But I agree it's fine to support the flip operation, I just realized it will not clutter the matrix toolbar anyway because the flip button is already in the global toolbar at the top. Honestly I'd just focus on the core functionality first and if there's time left, such nice-to-have features can still be added :) Up to you.

> The rotation is of the parts in the matrix, or relative to the path they would be placed on.
I don't think there are any options for setting the absolute or relative rotation of multiple parts.
besides editing the part parameters by hand for each one.

That's true, but actually it would make sense to provide that functionality in the select tool (in addition to "rotate", add a "set rotation angle" which just rotates each selected part individually).

And thanks for the image, rotation along a path is pretty cool :)

> While i did consider using the place devices side pane, it wouldn't be enough as is.
Ideally it should be possible to select placed or unplaced parts. But adding an option to arrange selected parts from
the list could be quite handy. (although that would require getting multiple selections working)

From the implementation point of view, I'd suggest anyway to provide the matrix tool only to parts which are already added to the board. The user has to go through the "add device" panel anyway to select the desired devices & footprints. This workflow still allows to implement an "Add devices in a matrix" later, which would just add the devices to the board and immediately invokes the matrix tool on them. So for the user it will be a smooth workflow while still keeping the implementation simple. I think a mix of placed and unplaced parts is not a real use-case and doesn't need to be supported.

> The way i think this that you'd open it with an existing selection either from tool list or context menu, or even place devices, or without one, and some way to select it.

Sounds good 👍 But I would really not support the use-case of starting the tool without a selection, it would just make the whole thing unnecessary complicated. And I think it is important to follow a consistent workflow (select objects, then apply operations) across the whole application.

Regarding implementation: Would you mind to implement the core logic of the placement in a dedicated class? I am thinking about something like this (just very rough pseudo-code):

```cpp
PlacementHelper helper;

// Set all options from the UI toolbar
helper.setMode(PlacementHelper::Mode::Matrix);
helper.setMatrixColumns(10);
helper.setHorizontalSpacing(UnsignedLength(1000000));

// Calculate the positions (TODO: and also return rotations)
const QList<Point> positions = helper.calcPositions(selectedParts.count());

// Update the position of selected parts
for (int i = 0; i < selectedParts.count(); ++i) {
  selectedParts[i].setPosition(positions[i]);
}
```

Such an implementation that doesn't depend on types like `Board` or `BI_Device` can easily be reused for other scenarios. For example in the footprint editor, and also for other kinds of objects (texts, circles, ...). And it will be easy to unit-test the behavior (especially important because this class will probably be extended with more features in future).

Just some idea, maybe you have already been thinking about a solution, maybe even better than this one :)

JustLike1984, 2025-11-27:
This is the current toolbar design, i've kept the rotation option for the path mode, as it already was quite short.

[image]
[image]

~ its about 520px width max for path mode, which means its smaller than the wire tool which is ~600px at minimum width.
I'm not sure where the code for the tool, is supposed to go. btw what is the deal with the commandset?