LibrePCB/LibrePCB #1600: Please provide a way to pass Rust crates cache to the build

yurivict, 2025-09-16:
Currently the build fails without internet connectivity.
Package builders don't allow internet connectivity for security reasons.
There should be a way to pass the CARGO_HOME environment variable to cargo that would contain the pre-downloaded crates.

ubruhin, 2025-09-16:
Doesn't this work out-of-the-box? I mean, you can set the `CARGO_HOME` environment variable and our build system should not need to care about it.

For the Flatpak we also do it that way: https://github.com/flathub/org.librepcb.LibrePCB/blob/aa27d87b4b8bf25a5d3d161d286edf0ea3036f29/org.librepcb.LibrePCB.yaml#L60-L63

What kind of package are you trying to create?

ubruhin, 2025-09-17:
Alright, I think it should work this way:

```
# Create a temporary cargo home
export CARGO_HOME=/tmp/cargo-home
mkdir -p $CARGO_HOME
cat > "$CARGO_HOME/config.toml" <<EOF
[source.crates-io]
replace-with = 'vendored-sources'

[source.vendored-sources]
directory = "$CARGO_HOME/vendor"
EOF

# Download crates (run this in the root dir of librepcb sources)
cargo vendor --manifest-path libs/librepcb/rust-core/Cargo.toml --locked "$CARGO_HOME/vendor"

# Build LibrePCB offline
ninja
```

Does that work? Unfortunately I did not (yet) find a simpler solution.:-/ Checking the vendored crates into our repository is not really practicable because that would be hundreds of megabytes (starting with LibrePCB 2.0, we will depend on Slint, which also requires a lot of crates). And the wrapping with CMake is required too as long as we have not fully migrated to Rust yet.

yurivict, 2025-09-17:
This doesn't work for me. Instead, I've proposed the enhancement in the FreeBSD that would support Rust subprojects.

----------

As it pertains to Rust, Rust is an inferior language to C++ because:
1. It lacks exceptions: this doesn't allow clean exit from complex contexts. Manual processing the Err value on every function return isn't really an option in complex algorithms since it requires a lot of manually written code.
2. Rust is a much less expressive language: Implementing same algorithms typically requires a lot more code in Rust. This creates opportunities for bugs: number of bugs is proportional to the amount of code.
3. No lambda functions: makes it difficult to implement polymorphism.

The promise of "safer code" is negated by the need to call unwrap() which panics the program.

-----------

In any case, I am closing this ticket since I already have a solution in my OS.

closed 2025-09-17 by yurivict